tenstorrent/tt-smi #1: Unable to get tt-smi working

eyonland, 2024-01-16:
python --version
Python 3.8.10

pip --version
pip 21.1.1 from /home/tickcode/.pyenv/versions/3.8.10/lib/python3.8/site-packages/pip (python 3.8)

tt-flash --fw-tar fw_pack-80.4.0.0_acec1267.tar.gz --force
Now checking device <builtins.PciChip object at 0x7f2bb77fd4b0>:
Looks like you are running a very old set of fw, assuming that it needs an update
Now flashing tt-flash version: (80, 4, 0, 0)
Forced ROM update requested. ROM will now be updated.

 tt-smi
 Unsupported FW version 1.1.0.0 detected on grayskull device 0.
 Require FW version >= 1.3.0.0 to run tt-smi
 Please update FW on device using tt-flash: https://github.com/tenstorrent/tt-flash

TTDRosen, 2024-01-16:
You need to reboot past flash, I'll update the output of tt-flash to make this clearer.

TTDRosen, 2024-01-24:
Closing as I think you have solved the problem

warthog9, 2024-01-24:
think this is closeable, if it's still broken re-open

closed 2024-01-24 by warthog9