tenstorrent/tt-smi #6: Allow specifying target filename for JSON config output

tt-rkim, 2024-02-16:
Right now, when using `-s`, it outputs a JSON with a pre-determined filename, appending a date.

This is not super great for automation. Could we have the ability to set a snapshot name just like the old SMI?

sbansalTT, 2024-02-20:
Hey Raymond, we have it already added that feature in, you have to use the `-f` flag to set the filename of the snapshot, exactly like old SMI. Link to instructions - https://github.com/tenstorrent/tt-smi?tab=readme-ov-file#snapshots

tt-rkim, 2024-02-21:
Hey @sbansalTT - I mean with the `-g` option, not the `-s` option.

For example, here, it would be nice if the first one generated `log.log` instead of a pre-determined JSON filename.

[image]

sbansalTT, 2024-02-21:
Oh gotcha - yes, I will add that into the next release

sbansalTT, 2024-03-11:
This has been fixed as of release 2.1.0
you should now be able to specify the file, or use the default with -g when generating the reset config

closed 2024-03-11 by TTDRosen