tenstorrent/tt-smi #7: tt-smi AICLK fixed on maximum freq

Dobob1022, 2024-02-28:
After running the demo code on grayskull, TT-SMI Said 
```AICLK(MHZ)
  1202/1202
```
the demo was ended but it's still fixed on 1202MHz. 

But I think it's temporary issue, Because After restart tt-smi it's shown as like normal. 

[image]

sbansalTT, 2024-03-11:
Hey, could you try again with the latest FW and latest SMI - improvements have been made and it would be helpful to see if it can be reproduced on the latest changes

warthog9, 2024-04-26:
This is probably related to tenstorrent/luwen#4

milank94, 2024-05-13:
@Dobob1022 following up to see if this is still an issue you are experiencing. Has the above guidance been helpful?

Dobob1022, 2024-05-14:
Sorry For Late reply, The firmware update and latest tt-smi can display like normal

sbansalTT, 2024-05-14:
Thanks! closing issue

closed 2024-05-14 by sbansalTT