tenstorrent/tt-smi #17: tt-smi and tt-tools-common have conflicting dependencies.

maychair, 2024-04-02:
1. build tt-smi with `pip install .` with error:
  ERROR: Cannot install tt-smi and tt-smi==2.0.0 because these package versions have conflicting dependencies.
  
  The conflict is caused by:
      tt-smi 2.0.0 depends on pyluwen 0.5.2 (from git+https://github.com/tenstorrent/luwen.git#subdirectory=crates/pyluwen)
      tt-tools-common 1.4.1 depends on pyluwen 0.5.2 (from git+https://github.com/tenstorrent/luwen.git@v0.3.6#subdirectory=crates/pyluwen)

2. And wh_reset.py in tt-tools-common move to tt_tools_common.reset_common, wh_resets.py import tt_tools_common.wh_reset

sbansalTT, 2024-04-02:
Hey Chao mei, thanks for your issue.
I think you might be running an older version of tt-smi that has old dependencies. 
Could you please uninstall your old smi, using `pip3 uninstall tt-smi` fetch the latest tt-smi, and then rebuild it with latest.

Do let me know if you still have any issues - then I can close this issue

maychair, 2024-04-10:
Hi @sbansalTT , i have checkout to tt-smi 47ca8f4 v2.2.0 and do `pip3 install . --ignore-installed`, some conflict shows up. here is the log:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
flax 0.6.0 requires rich~=11.1, but you have rich 13.7.0 which is incompatible.
pybuda 0.1.240401+dev.gs.c7ce1e7 requires requests==2.28.2, but you have requests 2.31.0 which is incompatible.
pybuda 0.1.240401+dev.gs.c7ce1e7 requires urllib3==1.26.14, but you have urllib3 2.2.1 which is incompatible.
tensorflow-cpu 2.13.0 requires typing-extensions<4.6.0,>=3.6.6, but you have typing-extensions 4.11.0 which is incompatible.
```
I use the lastest version pybuda-v0.10.9.gs.240401-alpha-ubuntu-22-04-amd64-python3.10.zip.

maychair, 2024-04-10:
I guess this pip error doesn't matter, v2.2.0 installed success and worked fine. You can close this issue now @sbansalTT and thanks for reply.

sbansalTT, 2024-04-10:
thanks! closing

closed 2024-04-10 by sbansalTT