tenstorrent/tt-smi #18: tt-smi is not running.

changmooklim, 2024-04-03:
I installed the Grayskull e75 on my computer and followed the procedure described in this link: "https://docs.tenstorrent.com/tenstorrent/add-in-boards-and-cooling-kits/grayskull-tm-e75-e150/software-setup".

After running tt-smi, I received the following message:  

Traceback (most recent call last):
  File "/usr/local/bin/tt-smi", line 5, in <module>
    from tt_smi import main
  File "/usr/local/lib/python3.8/dist-packages/tt_smi/__init__.py", line 7, in <module>
    from .tt_smi import *
  File "/usr/local/lib/python3.8/dist-packages/tt_smi/tt_smi.py", line 33, in <module>
    from tt_smi.tt_smi_backend import (
  File "/usr/local/lib/python3.8/dist-packages/tt_smi/tt_smi_backend.py", line 31, in <module>
    from tt_tools_common.utils_common.tools_utils import (
ImportError: cannot import name 'detect_chips_with_callback' from 'tt_tools_common.utils_common.tools_utils' (/usr/local/lib/python3.8/dist-packages/tt_tools_common/utils_common/tools_utils.py)

What should I check?

sbansalTT, 2024-04-03:
Hey Changmooklim, could you try the following and that should fix your issue - 
```
pip3 install --upgrade pip
pip3 uninstall tt-smi
pip3 install . --ignore-installed 
```
That should force the `tt_tools_common` library to upgrade to 1.4.1 which is needed for this function to work. 
Unfortunately you need a newer pip to solve this dependency

changmooklim, 2024-04-04:
tt-smi is working well. thank you very much.

closed 2024-04-04 by changmooklim