tenstorrent/tt-smi #24: TT-SMI textual breaks when moving the cursor too fast

hmohiuddinTT, 2024-04-29:
A customer is reporting an issue that breaks tt-smi when moving the cursor too fast: 

[image]

Probably a bug with textual? Let's see if there's any mention of it on the textual repo, or if there's a newer version of textual we can upgrade to.

Need to figure out how to repro this issue as well.

sbansalTT, 2024-05-15:
Hey Hammad, textual version has been bumped from 0.43.0 to 0.59.0 as of [v2.2.1](https://github.com/tenstorrent/tt-smi/commit/5fdfcea2afcb08ec8abfdca9b567b248eaeb647b)
Could you try it on your end and see if it fixes your issue?

warthog9, 2024-06-21:
since we haven't heard back on if this did/did not fix it, going to assume it's no longer an issue as we were never able to reproduce.  If it comes back up, reopen or file a new bug and we'll take another go at it.

closed 2024-06-21 by warthog9