tenstorrent/tt-smi #28: [Feature request] Show active PID and % utilization

adifrancescoTT, 2024-06-20:
Verbatim from Grayskull user:

"NVIDIA SMI tells you which PID is using their driver and % utilization; TT-SMI is nice looking but I don't get much information from it. I want to know which PIDs are using the card and the utilization %. It's hard to tell if the Tenstorrent card is doing anything"

Let me know if there is a better board to share this kind of feedback

sbansalTT, 2024-06-20:
Hey thanks for bringing it up. PID should be fairly straightforward to add, but the % util might be a bit harder to gather atm. Will consult with the team and update the issue when we have a plan to add that to SMI

joelsmithTT, 2025-10-16:
https://github.com/tenstorrent/tt-kmd/issues/145

% utilization is not a viable feature.