tenstorrent/tt-smi #30: AttributeError: 'FieldInfo' object has no attribute 'required'

dmakoviichuk-tt, 2024-07-12:
I installed latest version using pip install git+https://github.com/tenstorrent/tt-smi
and got this error.
```
Traceback (most recent call last):
  File "/home/ubuntu/.local/bin/tt-smi", line 5, in <module>
    from tt_smi import main
  File "/home/ubuntu/.local/lib/python3.8/site-packages/tt_smi/__init__.py", line 7, in <module>
    from .tt_smi import *
  File "/home/ubuntu/.local/lib/python3.8/site-packages/tt_smi/tt_smi.py", line 29, in <module>
    from tt_tools_common.reset_common.reset_utils import (
  File "/home/ubuntu/.local/lib/python3.8/site-packages/tt_tools_common/reset_common/reset_utils.py", line 13, in <module>
    import tt_tools_common.reset_common.host_reset_log as log
  File "/home/ubuntu/.local/lib/python3.8/site-packages/tt_tools_common/reset_common/host_reset_log.py", line 145, in <module>
    class PciResetDeviceInfo(ElasticModel):
  File "/home/ubuntu/.local/lib/python3.8/site-packages/tt_tools_common/reset_common/host_reset_log.py", line 57, in optional
    return dec(cls)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/tt_tools_common/reset_common/host_reset_log.py", line 50, in dec
    _cls.__fields__[field].required = False
AttributeError: 'FieldInfo' object has no attribute 'required'
```

it installed pydantic version 2.8.2. When I manually uninstalled it and installed pydantic version 1.2 error disappeared.

sbansalTT, 2024-07-15:
Thanks for reaching out Denys and informing us of the issue. 
I will try to replicate the issue on my end and push a fix in the next release

warthog9, 2024-07-15:
Ok I can't replicate this on my F40 box, but I'll admit I'm running out ahead on things there:
```
(venv) root@orko:~/tt-tools-common# python --version
Python 3.12.4
(venv) root@orko:~/tt-tools-common# pip --version
pip 24.1.2 from /root/venv/lib64/python3.12/site-packages/pip (python 3.12)
(venv) root@orko:~/tt-tools-common# pip show pydantic
Name: pydantic
Version: 2.8.2
Summary: Data validation using Python type hints
Home-page: https://github.com/pydantic/pydantic
Author: 
Author-email: Samuel Colvin <s@muelcolvin.com>, Eric Jolibois <em.jolibois@gmail.com>, Hasan Ramezani <hasan.r67@gmail.com>, Adrian Garcia Badaracco <1755071+adriangb@users.noreply.github.com>, Terrence Dorsey <terry@pydantic.dev>, David Montague <david@pydantic.dev>, Serge Matveenko <lig@countzero.co>, Marcelo Trylesinski <marcelotryle@gmail.com>, Sydney Runkle <sydneymarierunkle@gmail.com>, David Hewitt <mail@davidhewitt.io>, Alex Hall <alex.mojaki@gmail.com>
License: 
Location: /root/venv/lib64/python3.12/site-packages
Requires: annotated-types, pydantic-core, typing-extensions
Required-by: tt-smi, tt-tools-common
(venv) root@orko:~/tt-tools-common# cat /etc/redhat-release 
Fedora release 40 (Forty)
(venv) root@orko:~/tt-tools-common#
```

and things work fine for me there.  Can you get us some more information so we can try and figure out where the break is?

sbansalTT, 2024-07-17:
Hey Denys, I was able to repro your issue. The fix is to upgrade both tt-tools-common and tt-smi.
`pip3 install --upgrade https://github.com/tenstorrent/tt-tools-common.git`
and 
`pip3 install --upgrade https://github.com/tenstorrent/tt-smi.git`
Do let us know if the issue still persists

closed 2024-07-19 by dmakoviichuk-tt