tenstorrent/tt-smi #32: ModuleNotFoundError: Linkify enabled but not installed.

ihamer-tt, 2024-08-07:
To reproduce:
1. Download tt-smi from [as suggested in the online docs](https://docs.tenstorrent.com/tenstorrent/workstation-and-server-systems/t7000-4u-workstation/software-setup): `pip install git+https://github.com/tenstorrent/tt-smi`
2. Plain run it with `tt-smi` and pysh `h` for help

You get:
```
ModuleNotFoundError: Linkify enabled but not installed.
```

sbansalTT, 2024-08-07:
Hey Ivan, I can't seem to recreate the bug on any of my systems. I used the exact steps listed on 4 systems and they all worked.

Can you maybe try to upgrade your pip or use pip3 instead to download? 
SMI doesn't use `Linkify` for anything so it shouldn't be a requirement. 

Could you give me more details of your system?

ihamer-tt, 2024-08-07:
I am on bgd-lab-26
Docker container: ghcr.io/tenstorrent/tt-metal/tt-metalium/ubuntu-20.04-amd64:latest
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)

ihamer-tt, 2024-08-07:
On bgd-lab-t3002, it works (same docker), but installed in own python_env. On lab-26, I did not use a venv.

sbansalTT, 2024-08-07:
I'll debug on lab-26 without venv and get back to you

ihamer-tt, 2024-08-08:
It's most likely docker image related, rather than lab-26: ghcr.io/tenstorrent/tt-metal/tt-metalium/ubuntu-20.04-amd64:latest

marty1885, 2024-09-29:
I ran into the same issue when developing the tt-smi package for AUR. Please update the dependency list.

warthog9, 2024-10-30:
Ok doing pass through things @marty1885 @ihamer-tt do you have a way to verify that?  Running on the most minimal system I can, and running with the pyproject.toml to build, it's pulling in the dependencies the way I'd expect - but it looks like linkify is coming in through some dependency below ours, so I'm confused why it would be missing in any context.

Only thing I can think of is some hermetic build issue possibly?  @marty1885 was this reproducible on a minimal Arch system, I.E. if I can install one of those up quick and run an install should it fail?

marty1885, 2024-10-31:
@warthog9 I can try replicating in docker/VM later. But I'm expected to be quite busy till next weekend. May you poke me if I forgot about this?

warthog9, 2024-10-31:
@marty1885 no rush, I'll leave this open and if I don't hear anything in <random time period> I'll flag it back up :-)

closed 2025-04-30 by warthog9