tenstorrent/tt-smi #37: [grayskull 75] tt-smi returns error

vincent-kwon, 2024-08-16:
After inserting the card in PCI-3(instead of PCI-4 at this moment), and install the list software, I am getting error like below:

vincent@gamepc:~$ tt-smi
 Detected Chips: 1
 Detecting ARC: |
 Detecting DRAM: |
 [] ETH: |
Gathering Information ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--
Traceback (most recent call last):
  File "/home/vincent/.local/bin/tt-smi", line 8, in <module>
    sys.exit(main())
  File "/home/vincent/.local/lib/python3.8/site-packages/tt_smi/tt_smi.py", line 821, in main
    backend = TTSMIBackend(devices)
  File "/home/vincent/.local/lib/python3.8/site-packages/tt_smi/tt_smi_backend.py", line 80, in __init__
    self.smbus_telem_info.append(self.get_smbus_board_info(i))
  File "/home/vincent/.local/lib/python3.8/site-packages/tt_smi/tt_smi_backend.py", line 181, in get_smbus_board_info
    if pylewen_chip.as_bh():
AttributeError: 'builtins.PciChip' object has no attribute 'as_bh'

How should I fix it? is there any way to check the card installed and working normally?

sbansalTT, 2024-08-16:
Hello looks like your luwen library is out of date - reinstalling should fix that issue for you
`pip3 uninstall tt-smi -y && pip3 install .`
Please let me know if the issue still persists

vincent-kwon, 2024-08-16:
Thanks for quick support. Unfortunately, it didn't fix the issue. How can I install the latest luwen library?

milank94, 2024-08-19:
@vincent-kwon I would recommend trying to do a fresh install with the latest version of tt-smi following these steps:

Make sure you have rust installed:
```bash
sudo apt update
sudo apt install cargo
```

Intall tt-smi:
```bash
git clone https://github.com/tenstorrent/tt-smi.git
cd tt-smi
python3 -m venv .venv
source .venv/bin/activate
pip install -U pip
pip3 install .
tt-smi
```

closed 2024-08-28 by milank94