tenstorrent/tt-smi #39: Add support for snapshot to stdout

ihamer-tt, 2024-08-29:
It seems that we cannot use -s to just print out the information to stdout. This would be useful when coupled with other linux tools (such as grep. It should probably be the default. For example:

- `-s` print to stdout
- `-f file.json` save to file.json
- `-f` autogenerate the filename and save

tt-rkim, 2024-10-30:
An alternative is to immediately cat the file and use `jq` to parse JSON as an unblocker

TTDRosen, 2024-11-12:
@sbansalTT the intended workaround of `tt-smi -s -f /dev/stdout | jq` does not appear to work here. We might also want to consider making the print to stdout the default behaviour and let -f specify a file.

knauth, 2024-12-16:
Resolved by #59.

closed 2024-12-16 by knauth