tenstorrent/tt-smi #42: tt-smi appears to phone home serial numbers without user's consent or knowledge

afaerber, 2024-09-12:
The display of latest software versions appears to work via tt-tools-common getting a list of board IDs and querying cereal.tenstorrent.com for each board ID, i.e., serial number.

* Sending serial numbers home to the vendor seems a privacy concern. It should be opt-in with the user's awareness of what is being transmitted.
The `README.md` section "Latest SW Versions" doesn't say anything either.
* The `get_sw_ver_info` function's argument `show_sw_ver` is supplied in `tt_smi.py` from `get_latest_sw_vers = reactive(True)`, but there is no obvious way to toggle it.
* Even if there were a keyboard binding or checkbox, due to a bug in tt-tools-common it would still check online, just not return the obtained versions (tenstorrent/tt-tools-common#7).
* There is only one set of version numbers displayed in the bottom left box, even if I have two cards installed (cf. tenstorrent/tt-tools-common#8).
[image]
If a display per device were intended, both the API and the UI would need changes, such as adding a fourth tab or extending the FW Version tab with an actual / latest notation for the bundle and possibly tt-flash.
* Since the components all seem shared between cards and there being only one latest version for each component, it seems unnecessary to query for each card. A single query without `?SerialNumber=` argument could return those version numbers just as well.
Fake https://cereal.tenstorrent.com/?SerialNumber=42 works just fine, so there doesn't appear to be any actual check of the serial number currently (but requests likely get logged by the webserver).
In theory, the most hardware-specific component should be the kernel driver, which should go into mainline Linux at some point anyway and would then no longer apply to this list (cf. tenstorrent/tt-kmd#19).

smehtaTT, 2025-02-04:
Update - Patches in flight.

smehtaTT, 2025-02-25:
@warthog9 @sbansalTT - can you help close if done?

sbansalTT, 2025-02-25:
Sorry about not closing the issue. Yes, it has been patched as of - [v3.0.7](https://github.com/tenstorrent/tt-smi/tree/v3.0.7 ) with instruction on disabling in the readme.

closed 2025-02-25 by sbansalTT