tenstorrent/tt-smi #43: tt-smi --list shows Wormhole as resettable on Arm, whereas the UI says otherwise

afaerber, 2024-09-12:
On an Arm based Ampere Altra system, `tt-smi --ls` shows in the section "Boards that can be reset:" both Grayskull and Wormhole devices, whereas the tt-smi UI and tt-flash both say it's not supported for Wormhole on Arm. This seems inconsistent.

The list mode is probably lacking some check in `tt_smi_backend.py` function `print_all_available_devices`.
Device info would still be available above in section "All available boards on host:".

See also tenstorrent/tt-tools-common#6 for (lack of) background.

warthog9, 2024-10-31:
The answer is it works in some very specific instances it breaks, needing an Ampere system is part of that, having things setup on the Ampere exactly perfectly being another.  So the wording is inconsistent but not entirely wrong.  From the UI it's more a "chances are you don't have this setup", from the ls it's more a "you CAN try and reset it, the board supports reset" but it's not a guarantee that it'll work from an ARM system as a result of the number of things that need to be exactly right

knauth, 2025-02-03:
The new tt-smi UI doesn't show reset info; there are too many edge cases to give people good information. We are writing up some docs about this and will link them here when they're live.

sbansalTT, 2025-02-04:
https://github.com/tenstorrent/tt-smi?tab=readme-ov-file#resets
Updated the reset docs on the readme

knauth, 2025-02-05:
Closing as this is now resolved.

closed 2025-02-05 by knauth