tenstorrent/tt-smi #47: tt-smi does not fit a standard 80x24 terminal

afaerber, 2024-09-14:
The box "Compatibility Check" is cut off vertically in a default 80x24 GNOME terminal, hiding the "GS Reset" line and the "Latest SW Versions" box unless the terminal window is resized.

* Vertical screen space is wasted by a recommendation of Ubuntu 20.04+ (to users of other distros) that line-wraps unnecessarily, due to key alignment with the longest key "WH Reset".
[image]
* Driver check will always pass and could be dropped -- compare #45 checking for the driver in tt-smi `main()` vs. tt-tools-common `get_host_info()` always returning a string and thus `host_info["Driver"]` always evaluating as true.
* Memory is duplicated between "Host Info" and "Compatibility Check" boxes and could be merged as "Pass (x GB)".
* In "Host Info" maybe OS and Platform could be combined as, e.g., "Linux (aarch64)"?
[image]

knauth, 2024-12-20:
Hey @afaerber- we are working on some UI updates to make these widgets more compact and informative. You should see the first of those updates come in after the holidays. 

Until then, we're adding a compact mode (PR [64](https://github.com/tenstorrent/tt-smi/pull/64)) to hide the sidebar altogether and give more space for the main content.

knauth, 2025-02-03:
With the new host compatibility UI (PR [66](https://github.com/tenstorrent/tt-smi/pull/66)), this is resolved.

closed 2025-02-03 by knauth