tenstorrent/tt-smi #52: Add option to check if host-side reset needed

KarmaPoliceT2, 2024-10-24:
Use-case: User is using a vm with a card via pcie passthrough (kvm/qemu). We would like tt-smi to be able to do an "alive check" for a card and respond back on the status. By providing this we will allow users who may not have host-level access to know when to alert their administrators that intervention is required to get their card operational again. 

Ideally the intervention does not require rebooting the whole host (as other tenants may concurrently be using other cards on that system in their own vms)

TTDRosen, 2024-11-12:
Could you be more specific about what you see the workflow being? In general the easiest way to know if the host needs a reboot is to attempt a reset. If the reset fails then the next step would be to request for admin intervention. Are you looking for a way to check the status of the cards which doesn't require a reset to be issued?

warthog9, 2025-10-15:
closing this out since we haven't heard back, and there's been a lot of changes in the reset on the cards in the intervening time

closed 2025-10-15 by warthog9