tenstorrent/tt-smi #58: tt-smi indexing doesn't align with what's in /dev/tenstorrent/ and causing slight confusion on what needs to be passed where

warthog9, 2024-12-10:
Until such time as we have some other means of having a unique identifier for a card we should probably have the number that shows up in /dev/tenstorrent/ match the ID that tt-smi shows.

I.E.

```
# ls -lah /dev/tenstorrent/
total 0
drwxr-xr-x 2 root root     60 Dec  9 14:34 .
drwxr-xr-x 8 root root    400 Dec  9 14:34 ..
crw-rw-rw- 1 root root 236, 6 Dec  9 14:34 6
#
```

vs.

[image]

This will tend to lead folks to using `0` instead of `6` for things like reset.

warthog9, 2024-12-10:
Related to #56

hmohiuddinTT, 2025-02-20:
Related: #22

warthog9, 2025-04-28:
I'll note for anyone finding this: We shouldn't be doing anything based on any of the indexing.  The indexing, based on the /dev entry for example, is literally based on the order in which the kernel sees the device, so the numeric indexing could be wildly all over the place.  Likewise the indexing in this case inside tt-smi is a different number and a different ordering based on different assumptions and findings.

Realistically we need to shift away from numeric IDs, same way nics, physical drives, etc on Linux have all gone and if we have an index also provide serial number, pci path, etc for mapping (both in KMD and here)