tenstorrent/tt-smi #65: BH Telemetry shows garbage values for current and power

hmohiuddinTT, 2025-01-07:
[image]

sbansalTT, 2025-01-08:
Thanks for the issue - I'm guessing all the boards are on the latest 80.13.2.0 firmware. My guess is that there is an overflow on the telemetry values somewhere. 
Will take a look - I'll probably need access to your system, I haven't been able to re-create on my local boards

sbansalTT, 2025-01-15:
Hey Hammad, after looking into it seem like there is a manufacturing issue where some of the first batch of p150s have a power reporting issue. We are tracking this internally and it seems like your boards are from that batch.

hmohiuddinTT, 2025-02-07:
@sbansalTT is there any way to get around this issue via firmware on these boards?

warthog9, 2025-04-30:
Pretty sure is fixed

closed 2025-04-30 by warthog9