tenstorrent/tt-smi #68: Request for PCIE Slot ID in TT-SMI

hmohiuddinTT, 2025-02-19:
Cards locations are identified by which PCIe slot they are installed in on a motherboard. Currently there's no way to identify which slot a card is installed in on tt-smi. We work around this by mapping device_id (/dev/tenstorrent/*) -> PCIe bus id -> slot ID via the following command:

`sudo dmidecode -t slot`

It would be helpful to both cloud and customers if we can get this integrated into SMI.

hmohiuddinTT, 2025-02-19:
As an example the following ansible playbook prints the mapping:

```
---
- name: Check slot ids for boards installed on host
  hosts: compute
  gather_facts: false
  become: true
  tasks:
    - name: Get all board PCIe bus IDs
      ansible.builtin.shell: "lspci -d 1e52: | awk '{print $1}'"
      register: lspci_output
      changed_when: false

    - name: Get slot ID for each bus ID
      ansible.builtin.shell: dmidecode -t slot | grep -B8 {{ item }} | grep "Designation:" | awk -F ':' -F ' ' '{print $2}' | xargs
      register: slot_mappings
      loop: "{{ lspci_output.stdout_lines }}"
      when: lspci_output.stdout_lines | length > 0
      changed_when: false

    - name: Print slot IDs
      ansible.builtin.debug:
        msg: "{{ slot_mappings.results | map(attribute='stdout') | zip(lspci_output.stdout_lines) | map('join', ' | ') | join('\n') }}"
      when: lspci_output.stdout_lines | length > 0
```

hmohiuddinTT, 2025-02-19:
Please also add this info to the --list output along with bus id

warthog9, 2025-04-30:
There's a bunch of issues with us getting the slot from the motherboard itself, not the least of which the slot ids in dmidecode have a history of being sketchy / wrong.  We can have kmd and/or future health daemon potentially pull it in for consumption but I'd rather not depend on needing sudo there.

If we got a modification to where the stickers are at and get a QR code on the spine where we can scan the uid/serial number of the card would that suffice on tracking cards down?