tenstorrent/tt-smi #69: TT-QuietBox reboots during a tt-smi -r /path/to/reset_config.json

windwardly, 2025-03-08:
What would be useful for you to know?

In a TT-QuietBox that has otherwise seems to be running fine...

In a fresh build of `tt-smi` (commit 323ee254493b29556cd13220a46f732c5aafa7f8), this causes the computer to **_reboot_**:
```
(.venv) user@ub-22-04:~/user/repos/Huge/bootstrap$ tt-smi -g
 Detected Chips: 4
 Generated sample reset config file for this host: $HOME/.config/tenstorrent/reset_config.json 
 Update the generated file and use it as an input for the -r/--reset option. 

(.venv) user@ub-22-04:~/user/repos/Huge/bootstrap/tt-smi$ tt-smi -r $HOME/.config/tenstorrent/reset_config.json
 Starting PCI link reset on WH devices at PCI indices: 0, 1, 2, 3
```

From `tt-smi --snapshot_no_tty -f info`:
```
    "host_info": {
        "OS": "Linux",
        "Distro": "Ubuntu 22.04.5 LTS",
        "Kernel": "5.15.0-134-generic",
        "Platform": "x86_64",
        "Python": "3.10.12",
        "Memory": "503.43 GB",
        "Driver": "TT-KMD 1.31"
    },
    "host_sw_vers": {
        "tt_smi": "3.0.9",
        "pyluwen": "0.6.1"
    },
```

Aside, the number of chips detect changes from normal run to generate reset run:
```
$ tt-smi
Detected Chips: 8
```
```
$ tt-smi -g
Detected Chips: 4
[..[
```

sbansalTT, 2025-03-10:
Hey, thanks for your issue - could you pls try this and let us know what you are seeing?
/var/log/messages /var/log/syslog /var/log/kern.log from maybe 10 minutes before the reboot to after

windwardly, 2025-03-10:
Hi @sbansalTT ,
There is no `/var/log/messages`.
I ran the `tt-smi -r` at around `Mar  8 05:50`

I added newline to show before and after crash.

`/var/log/syslog` - shows finishing up firmware update. Then nothing. Then reboot:
```
Mar  8 05:40:00 ub-22-04 fwupdmgr[9842]: Downloading…: 97%
Mar  8 05:40:00 ub-22-04 fwupdmgr[9842]: Downloading…: 99%
Mar  8 05:40:00 ub-22-04 fwupdmgr[9842]: Idle…: 100%
Mar  8 05:40:00 ub-22-04 fwupdmgr[9842]: Idle…: 100%
Mar  8 05:40:00 ub-22-04 fwupdmgr[9842]: Successfully downloaded new metadata: 0 local devices supported
Mar  8 05:40:00 ub-22-04 systemd[1]: fwupd-refresh.service: Deactivated successfully.
Mar  8 05:40:00 ub-22-04 systemd[1]: Finished Refresh fwupd metadata and update motd.

Mar  8 05:54:54 ub-22-04 systemd-modules-load[823]: Inserted module 'msr'
Mar  8 05:54:54 ub-22-04 kernel: [    0.000000] Linux version 5.15.0-134-generic (buildd@lcy02-amd64-081) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #145-Ubuntu SMP Wed Feb 12 20:08:39 UTC 2025 (Ubuntu 5.15.0-134.145-generic 5.15.173)
Mar  8 05:54:54 ub-22-04 kernel: [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-134-generic root=UUID=691f9853-af33-4c00-b53c-4d2e7e28214b ro iommu=pt
```

`/var/log/kern` shows less:
```
Mar  8 04:58:56 ub-22-04 kernel: [   48.985116] audit: type=1400 audit(1741427936.226:52): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="snap.lxd.hook.install" pid=2082 comm="apparmor_parser"
Mar  8 04:58:56 ub-22-04 kernel: [   48.985275] audit: type=1400 audit(1741427936.226:53): apparmor="STATUS" operation="profile_replace" info="same as current profile, skipping" profile="unconfined" name="snap.lxd.hook.remove" pid=2083 comm="apparmor_parser"
Mar  8 04:59:00 ub-22-04 kernel: [   53.197905] loop7: detected capacity change from 0 to 8

Mar  8 05:54:54 ub-22-04 kernel: [    0.000000] Linux version 5.15.0-134-generic (buildd@lcy02-amd64-081) (gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #145-Ubuntu SMP Wed Feb 12 20:08:39 UTC 2025 (Ubuntu 5.15.0-134.145-generic 5.15.173)
Mar  8 05:54:54 ub-22-04 kernel: [    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.15.0-134-generic root=UUID=691f9853-af33-4c00-b53c-4d2e7e28214b ro iommu=pt
```

sbansalTT, 2025-03-18:
Hello, thank you for your patience. 
We made some updates on the luwen backend to help safe bail out when the chips are in a bad state. Could you pls pull and rebuild on the latest SMI and try it.
I would also suggest performing a cold boot (power off from the source) on the host system first. 
Pls let us know how it goes

windwardly, 2025-03-19:
Hi @sbansalTT . Built and tested: no change. Details:

I pulled my local `luwen` to 404896b74603c4e717184e6e33c63f7c2053267b
I edited my local `tt-smi` at b26e2e25990cb062dc8a9f0c6b7c02a551024a0a with:
`pyproject.tom` to fetch local `luwen`:
```
'pyluwen @ git+file:///<path-to>/luwen#subdirectory=crates/pyluwen',
```
and
```
--- a/tt_smi/tt_smi_backend.py
+++ b/tt_smi/tt_smi_backend.py
@@ -657,10 +657,12 @@ def pci_board_reset(list_of_boards: List[int], reinit=False):
                 CMD_LINE_COLOR.ENDC,
             )
             sys.exit(1)
+        del chip
 
     # reset wh devices with pci indices
     if reset_wh_pci_idx:
         reset_devices = WHChipReset().full_lds_reset(pci_interfaces=reset_wh_pci_idx)
+        del reset_devices
 
     # reset gs devices by creating a partially init backend
     if reset_gs_devs:
```
These change are to serialize the closing of the file handle and releasing of mmap regions when done. Otherwise, it is up to garbage collection causing overlaps in different parts of the code duplicating open resources for the same region.
Note: the first change "works" according to `strace`. The second change is never reached because reboot happens in `full_lds_reset()`.

I'm still debugging, but wanted to let you know that reboot happens with current `luwen` tip.

windwardly, 2025-03-19:
Current reboot isolation in `tt_tools_common/reset_common/wh_reset.py`
```
            # Trigger A3 safe state. A3 is a safe state where there are no more pending regulator requests.
            chip.arc_msg(self.MSG_TYPE_ARC_STATE3, wait_for_done=True)
            time.sleep(self.A3_STATE_PROP_TIME)
            # Triggers M3 board level reset by sending arc msg.
            if reset_m3:
                chip.arc_msg(self.MSG_TYPE_TRIGGER_RESET, wait_for_done=False, arg0=3)
            else:
                chip.arc_msg(self.MSG_TYPE_TRIGGER_RESET, wait_for_done=False)
            print(
                f"{CMD_LINE_COLOR.BLUE} Bottom of pci_chips loop {CMD_LINE_COLOR.ENDC}"
            )
            time.sleep(self.POST_RESET_MSG_WAIT_TIME)
```
Where I added the `print()` and `time.sleep()` at the end, I'm seeing `strace` output:
```
clock_nanosleep(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=1000000}, 0x7ffcbd37ef90) = 0
pselect6(0, NULL, NULL, NULL, {tv_sec=0, tv_nsec=30000000}, NULL) = 0 (Timeout)
pselect6(0, NULL, NULL, NULL, {tv_sec=2, tv_nsec=0}, NULL<EOF>
```

The first `time.sleep()` (traced as a `pselect6()`) completes and the second one doesn't. In old `straces`, I would see the `print()` show up as a `write()`. Not sure where they went. I'm running the code I think I am as that second sleep is not in the original. I believe `reset_m3` is false.

I'm out of debug time today. I'll keep hunting when I get back to it.

sbansalTT, 2025-03-19:
Thank you for your detailed debug trace! it is certainly helpful for us
May I ask what firmware do you have currently loaded up on your quietbox?

windwardly, 2025-03-19:
`80.13.2.0`

The output of `tt-smi` attached.

[tt-smi.json](https://github.com/user-attachments/files/19348085/tt-smi.json)

windwardly, 2025-03-19:
Deleted previous response - old log file.

`80.14.0.0`

[tt-smi.json](https://github.com/user-attachments/files/19348213/tt-smi.json)

sbansalTT, 2025-03-19:
Also when you get a chance, could you pls send a picture of the S/N sticker on the back? internally we want to trace the origin of the build

windwardly, 2025-03-20:
> [..] Change the SBIOS setting Chipset/AMD CBS/NBIO Common Options/NBIO RAS Common Options/PCIe Aer Reporting Mechanism to OS First.

We have a winner! That worked. Thank you!

Aret there any other SBIOS tweaks? 

> If the user loads default SBIOS settings, this setting will be lost.

Or, say, if [Google finds a security issue](https://github.com/google/security-research/security/advisories/GHSA-4xq7-4mgh-gp6w), [AMD issues a fix](https://www.amd.com/en/resources/product-security/bulletin/amd-sb-3019.html), and our intrepid almost-a-sysadmin attempts a BIOS upgrade... which upgraded to 1.0.0.D, but didn't fix the vulnerability, which will be coming soon from ASRock with 1.0.0.E .

 > EDIT: Confirmed that our system assembly partner should be configuring this properly for us, your system might have slipped through the cracks - so pls give it a try and let us know.

The machine was working fine for months. It was my figuring out how to do a BIOS upgrade that messed up the settings.

Since I'm likely to do a BIOS upgrade again someday, I want to have more detailed notes ready, especially since I lost any other custom settings that might have been done. Hence, what else do I need to configure?

If that's the only setting change, I'm ready to close. If there are others, I want to test them before closing.

sbansalTT, 2025-03-20:
Awesome! thanks for the update :) Thats the only SBIOS setting needed as far as I know - will update if there is anything else. 
If you upgrade again, you will need to re-apply the setting.
Will close for now - feel free to reach out or re-open if you have any more issues :)

closed 2025-03-20 by sbansalTT