tenstorrent/tt-smi #70: Discrepency Between Device Driver Reported Clocks and TT-SMI

SeanNijjar, 2025-03-12:
I am working on aus-glx-15 and am unable to determine the clock frequency on the (non-gateway) devices.

The top image is the log from device driver and the bottom is from tt-smi during a workload run
[image]

more info:

[image]

[image]

TTDRosen, 2025-03-14:
This appears to be a bug in the metal code https://github.com/tenstorrent/tt-metal/blob/ea570efb0a67aa8a729e54193faea1022f1d15d4/tt_metal/llrt/tt_cluster.cpp#L512 

I think I'm reading this correctly and seeing that the aiclk is being grabbed from the closest mmio device instead of the remote chip. Which I'm very confident, tt-smi is doing correctly

closed 2025-03-14 by TTDRosen

SeanNijjar, 2025-03-14:
Thanks Daniel 🐐