tenstorrent/tt-smi #72: Add 4U (and potentially 6U if it doesn't exist) on-board module firmware information to JSON output

tt-rkim, 2025-03-21:
Currently, tt-smi does not print out the firmware information of 4U Galaxy modules into the JSON output. 4Us will likely augment 6U stock for SW teams for a while.

It only prints out cards directly connected by PCIe. We have to go into the GUI screen to see everything.

I'm unsure if this is also the case for 6U, but if it is, we would like this to happen for 6U as well.

sbansalTT, 2025-03-21:
Yup this is a gap - will do, thanks for flagging!

knauth, 2025-04-01:
@tt-rkim We dove into this issue and it seems to be caused by a missing field in the firmware after flashing on Galaxy systems. We're working on a resolution for this, but for a quick fix, you can read from the tt_flash_version field instead- it's gettting the firmware version by mistake. See here:
```json
           "firmwares": {
                "fw_bundle_version": "N/A",
                "tt_flash_version": "80.16.0.0",
                "cm_fw": "2.31.0.0",
                "cm_fw_date": "2025-02-28",
                "eth_fw": "6.12.0",
                "bm_bl_fw": "N/A",
                "bm_app_fw": "N/A"
            },
```

tt-rkim, 2025-04-01:
Thank you!

Do you know which version of smi that's available from?

knauth, 2025-04-01:
This is latest; 3.0.12

knauth, 2025-04-01:
Note that four of the cards still have the correct formatting, like you noticed. You'll need to do a little bit of hacking there. We will have a better solution soon!

knauth, 2025-04-15:
Fixed as of release [v3.0.13](https://github.com/tenstorrent/tt-smi/releases/tag/v3.0.13). Let us know if there are any issues!