tenstorrent/tt-smi #73: Provide a package with tt-smi

dbabokin, 2025-04-02:
Installing `tt-smi` from source is not the most user friendly experience. Especially if the goal is to make it system-wide available.

Right now I do that with the following script:
```bash
temp_dir=$(mktemp -d)
pushd $temp_dir
git clone https://github.com/tenstorrent/tt-smi.git
cd tt-smi
sudo pip3.11 install pipx --root-user-action=ignore
sudo env PIPX_HOME=/opt/pipx env PIPX_BIN_DIR=/usr/local/bin pipx install .
popd
```

Note, I have to use `pipx`, as `tt-smi` has dependencies, which conflict with some other tools installed system-wide. I don't have anything special installed, but I see a conflict of `PyYAML` version. So I use `pipx` to have it virtualized.

Would be nice to be able to install it with just pip: `pip install tt-smi` or download package from Github Release.

sbansalTT, 2025-04-02:
Hey Dimitry, yeah packaging tt-smi is an ongoing effort and we are making some headway. @warthog9 FYI

blozano-tt, 2025-04-16:
We are doing this in CI, and it works well enough (It must be building the wheel from source tho).

```
pip install git+https://github.com/tenstorrent/tt-smi.git@v3.0.12
```

knauth, 2025-07-17:
@fahmedchanTT PR #124 is now open and should be reviewed soon.

warthog9, 2025-08-05:
package up on pypi and debs available from the release

closed 2025-08-05 by warthog9