tenstorrent/tt-smi #78: tt-smi Latest SW Versions box commonly confused with currently installed SW versions

tstescoTT, 2025-04-13:
Clearly good effort went into adding the tt-smi Latest SW Versions feature and making it informative for power-users, and I can see there is already support for optionally disabling it in https://github.com/tenstorrent/tt-smi/commit/68bb021ca6d13ed221225234c2b9e46695de16f0. I really appreciate this and would like to share some feedback and observations from seeing new developers use tt-smi in the wild and struggle with tt-smi and Latest SW Versions UI in particular that could help fix some common issues.

[image]

### Developer issues regarding tt-smi Latest SW Versions box
1. **Developers commonly interpret Latest SW Versions as current installed SW versions:** this has caused bad experiences for customers that needed to upgrade drivers or firmware and thought they already had the correct version, or did not "see the version change in tt-smi" after flashing firmware. This is the issue I see most commonly and main take away.
2. **Requesting data over internet for Latest SW Versions by default causes high startup latency**: I'm not sure if this is the root cause but anecdotally I see tt-smi start much faster when I set `"disable_sw_version_report": true` in `~/.config/tenstorrent/reset_config.json`. Also, for developers that are on-prem and cannot connect to the internet during runtime, could it be that if internet is unavailable then tt-smi startup time is increased because the request needs to timeout or come back with a connection error? I saw a developer today in this case wait ~2 minutes for `tt-smi` interface to startup multiple times during debugging.
3. **Developers may interpret Latest SW Versions as recommendation to upgrade SW:** I haven't seen someone do this yet but I'll raise it as another risk. This could cause confusion if the best practice is for developers to check for their hardware and target software stack support before upgrading.

I'd be happy to discuss possible solutions but I'd like to understand more about the intended use case for the Latest SW Versions before spending time on that.

milank94, 2025-04-14:
We can also remove TT-Buda from that list as it is no longer in active development.

smehtaTT, 2025-04-15:
@warthog9  to explore moving the "Latest SW Versions" to the firmware tab

knauth, 2025-04-15:
Hey folks- as of release [v3.0.13](https://github.com/tenstorrent/tt-smi/releases/tag/v3.0.13), we have removed the Latest SW Versions box. We are planning to add this functionality back in the next week or two with a better UX. Please let us know if there are any issues!

closed 2025-04-17 by milank94