tenstorrent/tt-smi #84: User Friendly Reset Prompting

bgrady-tt, 2025-04-28:
If the user attempts to reset a non-existent device (`tt-smi -r $NON_EXISTENT_DEV`), the tool crashes and barfs the stack trace at the end of this issue description.

This should at least be a short error message (`"tt-smi: Could not open the device at the index requested; are you sure this device exists?"`).

**_Ideally_** tt-smi would detect if the session is interactive TTY and _prompt the user to reset devices that do exist_:

```
Error: Cannot open device at index '0' for reset.

Reset one of the following devices instead? (y/n)
    Device ID: **3** Board Type: n300 L Board ID: .....
    Device ID: .....
Y
Which Device? (type index here)
3
You selected Device {something}, are you sure? (y/n)
Y
Running reset ....
```

**Backtrace example:**
```thread '<unnamed>' panicked at crates/pyluwen/src/lib.rs:559:70:
called `Result::unwrap()` on an `Err` value: DeviceOpenFailed { id: 0, source: Os { code: 2, kind: NotFound, message: "No such file or directory" } }
stack backtrace:
   0:     0x7f6d0fb072bb - std::backtrace_rs::backtrace::libunwind::trace::h3926e05c1d1f3b6d
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5
   1:     0x7f6d0fb072bb - std::backtrace_rs::backtrace::trace_unsynchronized::h9f5691494ac25ae6
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x7f6d0fb072bb - std::sys_common::backtrace::_print_fmt::h7e6bb7b81bf214f4
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x7f6d0fb072bb - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hcf688c88e28c91b4
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/sys_common/backtrace.rs:44:22
   4:     0x7f6d0fb3af00 - core::fmt::rt::Argument::fmt::h59a542682908b618
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/core/src/fmt/rt.rs:142:9
   5:     0x7f6d0fb3af00 - core::fmt::write::hce91e70849a27dee
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/core/src/fmt/mod.rs:1120:17
   6:     0x7f6d0fafd61d - std::io::Write::write_fmt::h0bba58d3b1b495e9
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/io/mod.rs:1762:15
   7:     0x7f6d0fb070a4 - std::sys_common::backtrace::_print::hf3a4f110a22f16df
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/sys_common/backtrace.rs:47:5
   8:     0x7f6d0fb070a4 - std::sys_common::backtrace::print::h0450d1fd5fc83f73
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/sys_common/backtrace.rs:34:9
   9:     0x7f6d0fb23dca - std::panicking::default_hook::{{closure}}::hee7ec73fab21a529
  10:     0x7f6d0fb23a6d - std::panicking::default_hook::he65be6b11b67d1e4
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/panicking.rs:292:9
  11:     0x7f6d0fb24108 - std::panicking::rust_panic_with_hook::h9e4f07a5a69c9caf
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/panicking.rs:779:13
  12:     0x7f6d0fb0769e - std::panicking::begin_panic_handler::{{closure}}::h69a9732dd2e7007d
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/panicking.rs:657:13
  13:     0x7f6d0fb074d6 - std::sys_common::backtrace::__rust_end_short_backtrace::hf159dc40d4738bc4
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/sys_common/backtrace.rs:170:18
  14:     0x7f6d0fb23e32 - rust_begin_unwind
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/std/src/panicking.rs:645:5
  15:     0x7f6d0fa50c75 - core::panicking::panic_fmt::hf38ef33e65607e17
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/core/src/panicking.rs:72:14
  16:     0x7f6d0fa51383 - core::result::unwrap_failed::h93afb55b612add5a
                               at /build/rustc-60UC9b/rustc-1.75.0+dfsg0ubuntu1~bpo0/library/core/src/result.rs:1653:5
  17:     0x7f6d0fa5965f - pyluwen::PciChip::new::h60840dced5646bf7
  18:     0x7f6d0fa81b8f - pyluwen::_::_::__INVENTORY::trampoline::hcb463decf805a0ad
  19:     0x7f6d1131716a - <unknown>
  20:     0x7f6d11376b2b - _PyObject_MakeTpCall
  21:     0x7f6d11142df3 - <unknown>
  22:     0x7f6d11147713 - _PyEval_EvalFrameDefault
  23:     0x7f6d11298e4b - _PyEval_EvalCodeWithName
  24:     0x7f6d11376124 - _PyFunction_Vectorcall
  25:     0x7f6d11142d6d - <unknown>
  26:     0x7f6d11147713 - _PyEval_EvalFrameDefault
  27:     0x7f6d1114e06b - <unknown>
  28:     0x7f6d11142d6d - <unknown>
  29:     0x7f6d1114446d - _PyEval_EvalFrameDefault
  30:     0x7f6d11298e4b - _PyEval_EvalCodeWithName
  31:     0x7f6d112991d2 - PyEval_EvalCodeEx
  32:     0x7f6d112995bf - PyEval_EvalCode
  33:           0x402e7b - <unknown>
  34:           0x403885 - <unknown>
  35:     0x7f6d1164fd90 - __libc_start_call_main
                               at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  36:     0x7f6d1164fe40 - __libc_start_main_impl
                               at ./csu/../csu/libc-start.c:392:3
  37:           0x401c4a - <unknown>
  38:                0x0 - <unknown>
Traceback (most recent call last):
  File "tt-smi.py", line 8, in <module>
  File "tt_smi/tt_smi.py", line 856, in main
  File "tt_smi/tt_smi_backend.py", line 638, in pci_board_reset
pyo3_runtime.PanicException: called `Result::unwrap()` on an `Err` value: DeviceOpenFailed { id: 0, source: Os { code: 2, kind: NotFound, message: "No such file or directory" } }
[PYI-25670:ERROR] Failed to execute script 'tt-smi' due to unhandled exception!```

knauth, 2025-04-30:
Duplicate of #56

warthog9, 2025-10-15:
@knauth can you double check if this is still an issue?