tenstorrent/tt-smi #88: tt-smi -r 0 fails on p100 (from a VM)

olofj, 2025-04-30:
I consistently get failures when trying to reset a p100a that's delegated to a VM (under Proxmox):

```
(.tenstorrent-venv) olofj@rvsw-bh-04:~$ tt-smi -r 0
 Starting PCI link reset on BH devices at PCI indices: 0
Waiting for up to 2 seconds for asic to come back after reset
 Config space reset not completed for device 0!
(.tenstorrent-venv) olofj@rvsw-bh-04:~$
```

This seems to be 100% reproducible.

Versions:

Firmware 80.18.0.0. BM App 0.3.1.0. tt-smi from current main branch.

warthog9, 2025-04-30:
Might need an updated bm firmware

olofj, 2025-04-30:
Yeah, I hit issues trying to update that. Filed internal JIRA SYS-1484 for that.

Another card in another VM on the same host works as expected, and that's on BM App 0.4.1.