tenstorrent/tt-smi #91: Should we add a --topology option?

blozano-tt, 2025-05-01:
For systems with known configurations, we could bypass the need for a reset_config.json ?

Should we add something like this?

Credit for idea goes to: @tt-rkim. 

```
tt-smi --help

OPTIONS
  --topology Topology of the system. One of: {single-card,t3k,tg,6u-galaxy,exabox...} 
```

sbansalTT, 2025-05-01:
Hey Bryan! Thanks for the suggestion. As long as we are supporting reset of complex configurations (TG) where the reset needs to be sequential for eth training + routing we will unfortunately need the reset_config.json
For just the PCIE cards, you can use the `-r` option to reset them.

closed 2025-05-01 by sbansalTT

tt-rkim, 2025-05-01:
@blozano-tt @asrinivasanTT and I agreed that for managing CI fleets:

- We will be using reset.json for topologies beyond single-card
- Metal infra will likely add an ansible role for copying the required JSON onto a machine, depending on its `topology`
- We will likely also add such scaffolding to provide the JSON on [CIv2](https://github.com/tenstorrent/github-ci-infra) so that jobs can use it and easily migrate.