tenstorrent/tt-smi #102: Failure of tools to function if installing in one Python venv

milank94, 2025-05-14:
Tools fail when installing in one Python virtual environment:

- tt-smi
- tt-flash
- tt-topology

Tested on Ubuntu 20.04 and 22.04, latest tool versions as of May 14.

Steps to reproduce:

1. Setup venv and install tt-smi

```bash
python3 -m venv test_env
source test_env/bin/activate
pip install -U pip
pip install git+https://github.com/tenstorrent/tt-smi
```

2. Run tt-smi

[image]

3. Install tt-flash

```bash
pip install git+https://github.com/tenstorrent/tt-flash.git
```

4. Run tt-smi

```bash
Traceback (most recent call last):
  File "/home/mkordic/test_env/bin/tt-smi", line 5, in <module>
    from tt_smi import main
  File "/home/mkordic/test_env/lib/python3.8/site-packages/tt_smi/__init__.py", line 7, in <module>
    from .tt_smi import *
  File "/home/mkordic/test_env/lib/python3.8/site-packages/tt_smi/tt_smi.py", line 29, in <module>
    from tt_tools_common.reset_common.reset_utils import (
ImportError: cannot import name 'ResetType' from 'tt_tools_common.reset_common.reset_utils' (/home/mkordic/test_env/lib/python3.8/site-packages/tt_tools_common/reset_common/reset_utils.py)
```

sbansalTT, 2025-05-14:
@knauth When you get some time, could you pls take a look? Thanks!

knauth, 2025-05-14:
Should be resolved as of latest flash.

closed 2025-05-14 by knauth

milank94, 2025-05-14:
Thanks @knauth ! Could you do a quick check on tt-topology as well to see if it needs the same fix?

knauth, 2025-05-14:
Topo was already on latest, so should be no issues there.