tenstorrent/tt-smi #112: tt-smi should report TBP draw as well as ASIC power draw

bgoelTT, 2025-06-09:
A lot of customers and developers need a way to measure the TBP draw. This is required for many reasons including but not limited to:
* performance comparison with competitor hardware
* in total system power budget calculations

Additionally, users expect feature parity with `nvidia-smi` https://developer.nvidia.com/system-management-interface which **does** report power draw at the board level.

This feature can be considered complete when we report TBP draw in `tt-smi` in the telemetry tab. Specifically we need the ability to query instantaneous, and average power draw via `luwen` and surfaced through `pyluwen`. The associated issue in that project is here:

sbansalTT, 2025-06-18:
@kikkiakki Do we have these values reported in telemetry table? If so it should be trivial to expose

bgoelTT, 2025-06-24:
@kikkiakki @sbansalTT is this scheduled to be worked on?

sbansalTT, 2025-06-24:
Best guess, in one of the next 2 release  - Priority is stabilizing reset for WH 6U galaxy in the upcoming release

bgoelTT, 2025-06-25:
@kikkiakki what is the correct field to query in the telemetry table? Also, is this surfaced via `pyluwen`? If all the information already exists there then we could start working on a small utility to help some of our customer testing

sbansalTT, 2025-06-25:
Yeah for the galaxy boxes that number will not be coming from the asic lvl telemetry - it would be from the BMC. 
Roping in @cyberDrake to give you details about that.
Asic lvl telemetry will still be from tt-smi

sbansalTT, 2025-06-25:
btw @bgoelTT You can get the telemtry struct dump if you do a `tt-smi -s` If the board power is there, it should be available via the json file anyway

kikkiakki, 2025-06-25:
the TBP tag still needs to be added to pyluwen. It won't show up in `tt-smi -s` until the tag is both added to luwen and exposed to tt-smi.