tenstorrent/tt-smi #116: tt-smi produces ~1.2MB of output during a `tt-smi -r0` on a p100a

olofj, 2025-06-19:
While on slow/sluggish internet, I noticed `tt-smi -r0` takes quite a while. Turns out that the spinning cursors when doing the rests produces a ton of output that normally isn't noticed on LAN connections, but when logged in over something like an in-flight wifi, it's noticable.

I.e:

```
olof@bh-vm:~/tt-bh-linux$ (. ~/.tenstorrent-venv/bin/activate ; tt-smi -r0) | wc
  34182  145306 1256302
```

It probably makes sense to make the tickers run a bit more slowly (i.e sleep a little between each iteration while waiting).

Cc: @sbansalTT @kikkiakki

warthog9, 2025-10-15:
Linked against https://github.com/tenstorrent/tt-smi/issues/150

kikkiakki, 2025-10-15:
Yeah, this should be fixed with the updates to no tty.