tenstorrent/tt-smi #118: Reset using tt-smi crashes Host

rganisettiTT, 2025-06-25:
Customer is using N300 wormhole and finds that the host crashes when reset using tt-smi. Following are the Hardware and software details:

Hardware:
—————
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Handle 0x0011, DMI type 1, 27 bytes
System Information
	Manufacturer: Penguin Computing
	Product Name: Relion XE2212GT-OEM
	Version: LCY2208LR5XX601
root@lum02# dmidecode -t processor
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Handle 0x001F, DMI type 4, 48 bytes
Processor Information
	Socket Designation: CPU0
	Type: Central Processor
	Family: Xeon
	Manufacturer: Intel(R) Corporation
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.
Handle 0x0012, DMI type 2, 17 bytes
Base Board Information
	Manufacturer: Intel Corporation
	Product Name: M50CYP2SBSTD
	Version: K42381-350

Software:
—————
{"Metalium": "0.59.0", "tt-smi": "3.0.20", "Firmware Bundle": "80.18.1.0", "tt-kmd": "2.0.0", "Buda": "0.19.3", "tt-flash": "3.3.3"}

warthog9, 2025-07-01:
Note: from a debugging perspective we've got a question out to the user since June 26th, specifically seeking clarification on what 'crash' does in this case.

Is it a full kernel stop type even or does it just spew things into the kernel log and run a kernel panic but the system stays up?

If it's the former: it's likely an issue with the way the system is handling the reset since our reset is a little messy (pcie device technically disappears and comes back on the system and more or less does a hotplug in place)

If it's the later we need a more comprehensive set of output to be able to start backtracking what's going on.

warthog9, 2025-07-02:
`lspci -vvv -d 1e52:` paste of that would be useful

But also have them check, similar to what @sbansalTT suggested,

https://www.intel.com/content/www/us/en/content-details/841362/intel-server-board-s2600-family-bios-setup-user-guide.html

page 123 specifically `4. PCIe AER Support`.  Should be set to `Enabled` as that looks to be the default, try flipping that to `Disabled`