tenstorrent/tt-smi #127: p150a fails initialization on Ampere Altra Max

RossComputerGuy, 2025-07-28:
I'm running NixOS 25.11 unstable on an Ampere Altra Max M128-26 with a Blackhole p150a. I see `tt-smi` fail like this:
```
Stage: DETECT
Detected Chips: 1
(4/5) ARC: BH FW boot not complete
 [8/8] DRAM

 [14/14] ETH
Traceback (most recent call last):
  File "/nix/store/50lycjl1fqnmfa90gnw4ih6sk261yq30-tt-flash-3.3.3/bin/.tt-flash-wrapped", line 9, in <module>
    sys.exit(main())
             ~~~~^^
  File "/nix/store/50lycjl1fqnmfa90gnw4ih6sk261yq30-tt-flash-3.3.3/lib/python3.13/site-packages/tt_flash/main.py", line 249, in main
    devices = detect_local_chips(ignore_ethernet=True)
  File "/nix/store/50lycjl1fqnmfa90gnw4ih6sk261yq30-tt-flash-3.3.3/lib/python3.13/site-packages/tt_flash/chip.py", line 320, in detect_local_chips
    for device in luwen_detect_chips_fallible(
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        local_only=True,
        ^^^^^^^^^^^^^^^^
    ...<2 lines>...
        noc_safe=ignore_ethernet,
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    ):
    ^
Exception: Chip initialization failed:
   Communication Status: Success
   DRAM Status: In Progress, 8 out of 8 initialized
   CPU Status: In Progress, 4 out of 4 initialized
   ARC Status: Timeout, 0 out of 1 initialized
   Ethernet Status: In Progress, 14 out of 14 initialized
   Noc Safe: true
   Unknown State: false
```

milank94, 2025-07-28:
@sbansalTT who would be the best person to have a look at this issue?

joelsmithTT, 2025-07-28:
Does this fix it?
```
joel@ampere ~/git/luwen $ git diff
diff --git a/crates/ttkmd-if/src/pci.rs b/crates/ttkmd-if/src/pci.rs
index 53da8db..32e90c2 100644
--- a/crates/ttkmd-if/src/pci.rs
+++ b/crates/ttkmd-if/src/pci.rs
@@ -543,7 +543,11 @@ impl PciDevice {
     /// This function requires that dest is a value returned by the self.register_address
     /// function.
     pub unsafe fn memcpy_from_device(dest: &mut [u8], src: *const u8) {
-        let align = core::mem::align_of::<u32>();
+        let align = if cfg!(target_arch = "aarch64") {
+            4 * core::mem::align_of::<u32>()
+        } else {
+            core::mem::align_of::<u32>()
+        };

         let mut offset = 0;
         while offset < dest.len() {
```
Tested like,
```
joel@ampere ~/git/luwen $ cd crates/pyluwen/
joel@ampere ~/git/luwen/crates/pyluwen $ make
<snip: it failed, but did produce a wheel before dying>
```
Go over to `tt-smi` venv, install the wheel:
```
(venv) joel@ampere ~/git/tt-smi $ pip install /home/joel/git/luwen/target/wheels/pyluwen-0.7.9-cp312-cp312-linux_aarch64.whl
<snip>
(venv) joel@ampere ~/git/tt-smi $ tt-smi # runs as expected, before I got bus error.
```

RossComputerGuy, 2025-07-29:
That code is already present so the patch doesn't work. https://github.com/tenstorrent/luwen/blob/v0.6.4/crates/ttkmd-if/src/pci.rs#L381-L385

RossComputerGuy, 2025-07-29:
Sweet, that patch fixes it.

[image]