tenstorrent/tt-smi #129: If board_id is N/A, then get_board_type throws a ValueError

kikkiakki, 2025-07-29:
We should just return N/A in this case.

```
(venv) user@quietbox-04:~/tt-tools/tt-smi$ tt-smi
 Detected Chips: 4
 Detecting ARC: \
 Detecting DRAM: \
 [] [14/14] ETH: \
Gathering Information ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━   0% -:--:--
Traceback (most recent call last):
  File "/home/user/tt-tools/tt-smi/venv/bin/tt-smi", line 8, in <module>
    sys.exit(main())
  File "/home/user/tt-tools/tt-smi/venv/lib/python3.8/site-packages/tt_smi/tt_smi.py", line 940, in main
    backend = TTSMIBackend(devices, pretty_output=is_tty)
  File "/home/user/tt-tools/tt-smi/venv/lib/python3.8/site-packages/tt_smi/tt_smi_backend.py", line 93, in __init__
    self.firmware_infos.append(self.get_firmware_versions(i))
  File "/home/user/tt-tools/tt-smi/venv/lib/python3.8/site-packages/tt_smi/tt_smi_backend.py", line 583, in get_firmware_versions
    get_board_type(self.get_board_id(board_num)) == "wh_4u"
  File "/home/user/tt-tools/tt-smi/venv/lib/python3.8/site-packages/tt_smi/tt_smi_backend.py", line 630, in get_board_type
    serial_num = int(f"0x{board_id}", base=16)
ValueError: invalid literal for int() with base 16: '0xN/A'
```

sbansalTT, 2025-07-29:
what board was this on?

kikkiakki, 2025-07-29:
@sbansalTT Some p300c in a Quietbox, which is in a bad state but that's another issue - we should probably be able to handle this case since board_id can be returned as N/A

vmilicevicTT, 2025-10-15:
@kikkiakki and @sbansalTT , can this be closed?  Thanks!

warthog9, 2025-10-15:
merged

closed 2025-10-15 by warthog9