tenstorrent/tt-smi #138: Add identify command to tt-smi

promed-sisaak, 2025-08-12:
Hi,

I would love a command line argument that lets us be able to identify the device corresponding to the PCI slot. 

My idea would be` tt-smi --identify n` which blinks the LEDs of the `n`'th device in order to visually identify the device.
`n` in this case is the # of the device in the tt-smi device enumeration.

[image]

CC: @bgoelTT

milank94, 2025-08-26:
@warthog9 Can you please review and assign someone.

warthog9, 2025-10-15:
personally I'd just want to set "ident" and let the MCU go blink the LED, likely with some sort of a timeout (an hour?) or something.