tenstorrent/tt-smi #151: [Bounty $2500] Bring up a Rust clone of TT-SMI

tsingletaryTT, 2025-09-22:
# Bring up a rust clone of TT-SMI — $2500, Hard

**Project/Repo:** [TT-SMI](https://github.com/tenstorrent/tt-smi)  
**Category:** Feature

---

## 📝 Background
Migration of the Tenstorrent Tooling suite (SMI, Flash, Topology, etc.) to Rust would ease packaging and distribution, reduce technical debt, and improve developer and customer experience. As a first step, we would like help migrating TT-SMI, our system management interface, to a native Rust project.

## 🎯 What Success Looks Like
- tt-smi-rs GitHub project created and made public
- Apache 2 license applied to project
- Project can be built with cargo
- Project depends on Luwen, our library for high-level TT device access
- After build, project achieves CLI/GUI feature parity with existing tt-smi

## 💡 Problem to Solve
This problem involves porting tt-smi to rust with feature parity. You will need to duplicate all GUI and CLI features that we currently have while keeping a clean, maintainable project structure.

## 🧭 Guidance & Starting Points
- [Luwen](github.com/tenstorrent/luwen)
- Ask June Knauth, John Hawley, and Samridhi Bansal for help

---

## 📚 Resources
- [TT-SMI](github.com/tenstorrent/tt-smi)

Gogopex, 2025-09-22:
heya, I'd like to claim this one! I already have a very solid base here https://app.radicle.xyz/nodes/seed.radicle.garden/rad%3Az27degs9voTdJQb9XauJz6RY4Mjk9 - will port it over to GitHub, do a comparative analysis of tt-smi and tt-smi-rs and send over the repo here!

Gogopex, 2025-09-22:
Here: https://github.com/Gogopex/tt-smi-rs

>tt-smi-rs GitHub project created and made public
✔️
>Apache 2 license applied to project
✔️
> Project depends on Luwen, our library for high-level TT device access
✔️

> Project can be built with cargo
> After build, project achieves CLI/GUI feature parity with existing tt-smi

Will ensure these two are ✔️ before pinging again

tsingletaryTT, 2025-09-22:
@Gogopex Glad you're up for this rather inspired bounty! We will get you officially assigned as soon as possible; just some administrative duties ahead of that. In the meantime, consider yourself first in line and that the [Bounty Terms & Conditions](https://docs.tenstorrent.com/bounty_terms.html) should be read and understood before proceeding much further. I'll get more official in a follow up comment!  Thanks for being eager to contribute!

Gogopex, 2025-09-24:
Thank you @knauth! I recently moved out and I'm getting internet tomorrow morning at my new place so I'll be able to iron out all the kinks you listed.

Gogopex, 2025-10-05:
2 weeks later: I finally have a new home and internet and everything setup, so looping back to this.

@knauth any news on luwen crates being updated? Can I help? I switched the Rust edition to 2021 since I didn't actually use any unstable features. In the meantime, either we can clone it as a submodule and make sure it's built properly with a small script, or I can mess around with a cargo workspace.

closed 2025-11-04 by jberkowitzTT